nootropicdesign/lora-mesh #2: Limited amount of nodes in mesh network?

vannimb, 2019-06-13:
Hello,
I see in LoRaMesh.ino the following lines:
  nodeId = EEPROM.read(0);
  if (nodeId > 10) {
    Serial.print(F("EEPROM nodeId invalid: "));
    Serial.println(nodeId);
    nodeId = 1;
  }

does this means that the mesh network size is limited to 10 units maximum including the gateway?

If this is the case, is there a reason for this limit?

noeldum, 2020-05-04:
I second the question. What is the max number of nodes in Radiohead?

nodeid is defined as uint8_t. Does this means we are limited to 256 nodes?

nootropicdesign, 2020-05-04:
This particular demonstration supports 10 nodes, but I've only tried 4. More nodes means constant collision on the airwaves, depending on how often a node transmits. The RadioHead library supports only 255 nodes, per the 8 bit value. I don't think 0 is a valid node number.

closed 2020-05-08 by nootropicdesign