nootropicdesign/lora-mesh #3: Compiling in esp32

jpunn, 2019-07-01:
Arduino: 1.8.5 (Linux), Board: "Heltec_WIFI_LoRa_32, 80MHz, 921600"

WARNING: Category 'Language' in library ArduinoStreaming is not valid. Setting to 'Uncategorized'
Archiving built core (caching) in: /tmp/arduino_cache_366318/core/core_espressif_esp32_heltec_wifi_lora_32_FlashFreq_80,UploadSpeed_921600_808b03452edbafa9ebcae00cb952e117.a
sketch/Router.ino.cpp.o:(.literal._Z7freeMemv+0x0): undefined reference to `__brkval'
sketch/Router.ino.cpp.o:(.literal._Z7freeMemv+0x4): undefined reference to `__heap_start'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "RHRouter.h"
 Used: /home/jpy/Arduino/libraries/RadioHead-master
 Not used: /home/jpy/Arduino/libraries/RadioHead
 Not used: /home/jpy/Arduino/libraries/RadioHead
 Not used: /home/jpy/Arduino/libraries/RadioHead
 Not used: /home/jpy/Arduino/libraries/RadioHead
exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

HI
I TRIED TO RUN LORA MESH IN ESP32 AND END UP WITH THIS ERROR. PLS HELP ..

pablocaol, 2019-11-03:
hello I also have the same mistake you could solve it

fmc5046, 2020-07-13:
Hi I had this  problem too with the ESP32

prlombaard, 2020-08-15:
@pablocaol / @fmc5046 i got the same error, i had to remove the freemem function completely from the code to get it compiled. But after compilation i have my ESPs have a kernel panic (GuruMeditation). It happens as soon as the updateRoutingTable function is called. if i remove the call to UpdateRoutingTable function then the loop() runns OK, obviously not worth anything because routing table not displayed...

badaniels, 2020-10-01:
@prlombaard did you ever manage to rectify this problem? I have the same issue. The exact function that it fails on is `routes[n-1] = route->next_hop`

I cant seem to find where this pointer is referencing, which is possibly why its crashing. If you did rectify it I would love to know.

Thanks

GregorR1, 2020-10-16:
I'm also getting this same issue.  I'm attempting to use this on a TTGO T-Beam which is ESP32 based.  I have hardcoded the nodeId as 2 on one of my boards to test.  It appears that the line "manager = new RHMesh(rf95, nodeId);" isn't initialising as I'm getting the "init Failed" warning.  Could this be part of the issue?

enrico1036, 2020-12-16:
Compilation error `(.literal._Z7freeMemv+0x0): undefined reference to __brkval' `, `(.literal._Z7freeMemv+0x4): undefined reference to __heap_start'`
could be resolved changing `freeMem()` function to 
```
int freeMem() {
  return ESP.getFreeHeap();
}
```

vibeeshanan, 2021-02-20:
#elif RH_TEST_NETWORK==3
where we put this command?

vibeeshanan, 2021-02-20:
how does the board create a table in the first stage?

nootropicdesign, 2021-02-20:
If you want to use a test network, define the constant in your code so that the Radiohead library knows which network topology to use.

`#define RH_TEST_NETWORK 3`

Please study the code and Radiohead before you post comments here. These are not defects in my project.

fnasarias, 2021-02-24:
@badaniels I also cannot get it to work. Removing the line `routes[n-1] = route->next_hop` prints the table but table does not update. Have you resolved this issue? Would appreaciate some help here. Thanks!

vibeeshanan, 2021-02-24:
when we transfer the message from one node to another node how to send that message using this mesh code?. I don't see the message any message send code in the mesh code.

nootropicdesign, 2021-02-26:
This code demonstrates the mesh networking implemented in RadioHead. This code does not implement mesh networking. To send a message through the mesh, use sendToWait, just like this demo shows.

vibeeshanan, 2021-03-03:
what kind of algorithm did you use to get the routing table?

nootropicdesign, 2021-03-03:
RadioHead implements the mesh networking. It sends route discovery messages to determine how nodes are connected. If you want to understand mesh networking, study the RadioHead library.

JoseCarTorBel, 2021-03-08:
Hello, when I execute this on ESP32 TTGO, I have the next problem:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
initializing node done
RF95 ready
mem = 354672
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1051  PS      : 0x00060f30  A0      : 0x800d1176  A1      : 0x3ffb1f40  
A2      : 0x00000002  A3      : 0x3ffc038d  A4      : 0x3ffc0386  A5      : 0x3ffc0250  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000002  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000002  A12     : 0x0000000a  A13     : 0x00e4c000  
A14     : 0x7ff00000  A15     : 0x7ffc9800  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400ed53c  LEND    : 0x400ed559  LCOUNT  : 0x00000000  

Backtrace: 0x400d1051:0x3ffb1f40 0x400d1173:0x3ffb1f60 0x400d571d:0x3ffb1fb0 0x40088db5:0x3ffb1fd0

Someone know how to fix?

prlombaard, 2021-06-03:
> @prlombaard did you ever manage to rectify this problem? I have the same issue. The exact function that it fails on is `routes[n-1] = route->next_hop`
> 
> I cant seem to find where this pointer is referencing, which is possibly why its crashing. If you did rectify it I would love to know.
> 
> Thanks

Not sure if you had any luck again, sorry i dropped this project, life happened, will have to fire up the things again to have a look. Maybe the latest release of RadioHead fixes this issue.

dougie181, 2021-09-29:
When f

> Hello, when I execute this on ESP32 TTGO, I have the next problem:
>
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee [...]

Yes... Check if (route == NULL) and (route->next_hop == NULL) before attempting to access them. If they are NULL, then do something sensible :-)

Thomasalex2, 2022-08-01:
Has anyone managed to compile the LoraMesh ino file on ESP32 successfully?

davidefa, 2022-11-02:
Hope not to sound inappropriate... but if you want a 'super basic' example of the radiohead mesh library you can follow this example: https://www.hackster.io/davidefa/esp32-lora-mesh-1-the-basics-3a0920 ( for ttgo/heltec lora boards )